TomNisbet/TommyPROM #2: Shift registers

coolname1, 2019-02-10:
Hi there,

Not an issue as such... Can this project use the 74HC595 shift registers as an alternative shift register? I have about 20 of those, but no 74LS164. Thanks :)

TomNisbet, 2019-05-16:
Sorry for the late reply.  Any serial-in parallel-out shift registers should work, although there would be a few pin changes.  For example, the 72HC595 has a OE that would just be tied to ground to have it always enabled.

closed 2019-05-16 by TomNisbet

OmaiGrundles, 2020-03-15:
I would like to mention that to get the 74HC595 to work, I had to add code to PromAddressDriver.cpp: defining an unused pin(10), and toggling it high then low at the end of PromAddressDriver::setAddress. Then attaching pin 10 from the arduino to the RCLK pin of each shift register.

I am a complete newbie when it comes to this kind of stuff so there probably is a better way though.

I would also like to thank @TomNisbet for doing all this. You saved the project I was working on, I had no other way of getting the darn write protection off of my AT28c256's! People like you are amazing! :)

TomNisbet, 2020-03-17:
I just looked at the data sheet and saw the two stage design for that chip.  Your addition to clock the data from the shift register to the storage register makes perfect sense.  Glad it is working for you.

ABelliqueux, 2024-05-29:
@OmaiGrundles : Could you share your modified code here ? I plan to use 74HC595s too...

ABelliqueux, 2024-05-29:
Oh I didn't realize these instructions applied to 74HC too. Thanks for the clarification !

ABelliqueux, 2024-05-31:
Last question : Is the TommyPROM32 PCB compatible with 74HC595 too ? I suspect it needs modification to connect the RCLK pins ?

ABelliqueux, 2024-06-01:
Just checked the PCB quickly and noticed the tommyprom32 is designed around two 74HCT164s that don't have pinouts similar to 74LS/74HC : 
[image]
hence the notes [here](https://tomnisbet.github.io/TommyPROM/docs/hardware#74ls595-shift-registers) wont apply. 

Would adapting the TommyProm32 PCB to 74HC/LS require a lot of modifications to the PCB ?

ABelliqueux, 2024-06-02:
Hi, I think I'll give it a try if it's not too difficult. I'll check the datasheets for these components and hopefully I'll be able to figure out how to alter the layout. I'll keep you updated here if you don't mind the thread hijacking too much :sweat_smile:

TomNisbet, 2024-06-02:
Looking at your schematic above, the good news is that the schematic symbol pins for the outputs of the 595 are in the same place as the 164.  You should be able to delete the control wires near the chip on the left side and then slide the 595 into place with QA connected to A0 and QH connected to A7.  Then you just need to connect the appropriate control signals on the left.

After you delete the 164s, number the two 595s as U1 and U2.  You will also need to assign the DIP-16 footprints to the 595s in the schematic editor.  You can then go the the PCB layout and use the Update From Schematic button to replace the 164s with the 595s.  I deleted the wire segments closest to the chip to have some space to work.  It looks like the layout may be easier if you rotate the chips so that pin 1 is at the bottom.

[image]

ABelliqueux, 2024-06-03:
Thanks for your pointers !
Here is what I've come up with so far : 

[image]

according to : 

[image]
55/44bfa90c-456e-4fe3-82db-d7296dde34c9)

I also removed J12 and connected D13 to LSR+USR RCLK directly as per [the doc](https://tomnisbet.github.io/TommyPROM/docs/hardware#74ls595-shift-registers). 

I'm not sure about pins 10 and 13 as their name differ between the datasheet and kicad's libary ( MR/OE for the 164 vs SRCLR/OE for the 595 ).

TomNisbet, 2024-06-04:
Yes. SRCLR is high and OE is low.

ABelliqueux, 2024-06-05:
Here goes ... I hope I didn't make mistakes, the DRC tests pass fine but for one warning about the "+5V" marking overlapping something :  

[image]

A vector version is available here for higher resolution : 

[TommyPROM32-brd.pdf](https://github.com/user-attachments/files/15592335/TommyPROM32-brd.pdf)

And the pro file is here : 

[TommyPROM32-2024-06-05_163940.zip](https://github.com/user-attachments/files/15592380/TommyPROM32-2024-06-05_163940.zip)

ABelliqueux, 2024-06-14:
Just a heads up to say that the breadboard version of the schematics for the TP32-595 version is working as intended. I think I'll order some pcbs soon and confirm all is well here.

ABelliqueux, 2024-07-29:
Well, here it is, working all right AFAICT (did several reads/dumps checking checksum), the 595 version :)

[image]