TomNisbet/TommyPROM #5: TommyProm not running when connecting to Nano

8BitMark, 2020-05-02:
When connecting to the terminal I get the > prompt but then nothing further, on occasion I've got the TommyProm v2 & instructions message when pressing enter, but there is a very long wait.  Then nothing further works.

I've been using 74HC595s using wiring based on Ben Eater's EEPROM breadboard project and the fork of your project from leomil72.  I've double-checked my wiring and I can get Leo's modified version of TommyPROM to run but it writes the same value to all addresses :-(   Any ideas what I've gone and messed up?

MarkC0x, 2020-05-16:
Hi @8BitMark Did you figure this out by any chance? I am also using two 74HC595’s

I think I am in the same situation as you.  In my case the Zap test fails to verify but if I repeatedly run D0000 to dump out the first 255 bytes it will eventually show the correct data. It’s very strange but I figure it must be something like a timing issue on the read side.  Does your data eventually read correct after retries?

8BitMark, 2020-05-16:
I added some delays to the write code and it's been working for me fine
since.  I also had some incorrect wiring so that plus the delays fixed it.
I'm not near my computer at the moment but will share details later.

TomNisbet, 2020-05-17:
I've never tried using 595's.  I had 74LS164s in my parts bin, so that's what's in the design.  A lot of people are using the 595 because of Ben Eater's design, so I'll pick up a few next time I order chips and will give them a shot.

With that said, there's no reason it shouldn't work with them, When you say that the D0000 command eventually works, what is the output on the runs where it doesn't work?

MarkC0x, 2020-05-18:
Hi @TomNisbet, thanks for the reply.  The data on the `D0000` command in the intermittent outputs (before the correct output) seems somewhat random to me, although a lot of` 0x1a` appear.  Here is a sample run output (with timestamps on the Arduino serial monitor):

```
19:21:33.978 -> >d0000 007f
19:21:35.191 -> 0000: ff ff ff ff  ff ff ff ff  ff ff ff ff  ff ff ff ff  ........ ........ 
19:21:35.191 -> 0010: ff ff ff ff  ff ff ff ff  ff ff ff ff  ff ff ff ff  ........ ........ 
19:21:35.224 -> 0020: ff ff ff ff  ff ff ff ff  ff ff ff ff  ff ff ff ff  ........ ........ 
19:21:35.224 -> 0030: ff ff ff ff  ff ff ff ff  ff ff ff ff  ff ff ff ff  ........ ........ 
19:21:35.224 -> 0040: ff ff ff ff  ff ff ff ff  ff ff ff ff  ff ff ff ff  ........ ........ 
19:21:35.224 -> 0050: ff ff ff ff  ff ff ff ff  ff ff ff ff  ff ff ff ff  ........ ........ 
19:21:35.224 -> 0060: ff ff ff ff  ff ff ff ff  ff ff ff ff  ff ff ff ff  ........ ........ 
19:21:35.258 -> 0070: ff ff ff ff  ff ff ff ff  ff ff ff ff  ff ff ff ff  ........ ........ 
19:21:35.258 -> 
19:21:35.258 -> >z0000
19:21:39.927 -> 
19:21:39.927 -> ERROR: Verify failed  addr=0x0  read=0x1A  expected=0x41
19:21:39.962 -> 
19:21:39.962 -> >d0000 007f
19:21:43.668 -> 0000: 1a 1a 1a 1a  5a a2 1a 1a  1a 1a fb 5d  1a 1a 08 f0  ....Z... ...].... 
19:21:43.668 -> 0010: 1a 1a 1a 1a  1a 1a 1a 1a  2f 1e c1 1e  1a 1a 1a 1a  ........ /....... 
19:21:43.668 -> 0020: 1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  ........ ........ 
19:21:43.668 -> 0030: dc 25 12 da  1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  .%...... ........ 
19:21:43.705 -> 0040: 1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  ........ ........ 
19:21:43.705 -> 0050: 1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  ........ ........ 
19:21:43.705 -> 0060: 1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  ........ ........ 
19:21:43.705 -> 0070: 1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  ........ ........ 
19:21:43.741 -> 
19:21:43.741 -> >d0000 007f
19:21:46.803 -> 0000: 1a 1a 1a 1a  5a a2 1a 1a  1a 1a fb 5d  1a 1a 08 f0  ....Z... ...].... 
19:21:46.803 -> 0010: 1a 1a 1a 1a  1a 1a 1a 1a  2f 1e c1 1e  1a 1a 1a 1a  ........ /....... 
19:21:46.836 -> 0020: 1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  ........ ........ 
19:21:46.836 -> 0030: dc 25 12 da  1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  .%...... ........ 
19:21:46.836 -> 0040: 1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  ........ ........ 
19:21:46.836 -> 0050: 1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  ........ ........ 
19:21:46.836 -> 0060: 1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  ........ ........ 
19:21:46.871 -> 0070: 1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  1a 1a 1a 1a  ........ ........ 
19:21:46.871 -> 
19:21:46.871 -> >d0000 007f
19:21:53.355 -> 0000: 12 0a 0a 0a  5a a2 1a 1a  1a 1a 3b 5d  1a 1a 08 f0  ....Z... ..;].... 
19:21:53.393 -> 0010: 1a 1a 1a 1a  1a 1a 1a 1a  2f 1e c1 1e  1a 1a 1a 1a  ........ /....... 
19:21:53.393 -> 0020: 1a 1a 1a 1a  1a 1a ff ff  1f ff ff ff  ff fa ff ff  ........ ........ 
19:21:53.393 -> 0030: dc 2f 12 da  1a ff 1f ff  1a ff 1a ff  1a ff ff ff  ./...... ........ 
19:21:53.393 -> 0040: 9a da da ff  ff ff ff ff  ff ff ff ff  ff ff ff ff  ........ ........ 
19:21:53.393 -> 0050: ff ff ff ff  ff ff ff ff  ff ff ff ff  ff ff ff ff  ........ ........ 
19:21:53.393 -> 0060: ff ff ff bf  ff ff ff fa  fa ff ff ff  ff ff ff ff  ........ ........ 
19:21:53.424 -> 0070: ff ff ff ff  ff ff ff ff  ff ff ff ff  ff ff ff ff  ........ ........ 
19:21:53.458 -> 
19:21:53.458 -> >d0000 007f
19:22:02.620 -> 0000: 41 42 43 44  45 46 47 48  01 02 04 08  10 20 40 80  ABCDEFGH ..... @. 
19:22:02.620 -> 0010: 7f bf df ef  f7 fb fd fe  00 ff 55 aa  30 31 32 33  ........ ..U.0123 
19:22:02.620 -> 0020: ff ff ff ff  ff ff ff ff  ff ff ff ff  ff ff ff ff  ........ ........ 
19:22:02.620 -> 0030: ff ff ff ff  ff ff ff ff  ff ff ff ff  ff ff ff ff  ........ ........ 
19:22:02.658 -> 0040: ff ff ff ff  ff ff ff ff  ff ff ff ff  ff ff ff ff  ........ ........ 
19:22:02.658 -> 0050: ff ff ff ff  ff ff ff ff  ff ff ff ff  ff ff ff ff  ........ ........ 
19:22:02.658 -> 0060: ff ff ff ff  ff ff ff ff  ff ff ff ff  ff ff ff ff  ........ ........ 
19:22:02.658 -> 0070: ff ff ff ff  ff ff ff ff  ff ff ff ff  ff ff ff ff  ........ ........ 
19:22:02.692 -> 
```

MarkC0x, 2020-05-19:
I just wanted to circle back on this, and let folks know that I solved my issue.

As it turns out - and probably unsurprisingly - it was human error on my part.  I am using the AT28C256 EEPROM, and I neglected to properly connect the additional (unused) address lines to ground (hence leaving them floating).  That explains the strange reads which would eventually become correct.

Feeling somewhat embarrassed, so I'll leave by the side entrance now :-)

TomNisbet, 2020-05-19:
I was just about to reply that the write was clearly working because it eventually returned the correct data on the read.  Would have suggested a short delay in PromDevice28C before the readData to let everything settle.  But your explanation makes a lot more sense!

Glad you got it working.  Why do you have unused address lines?  My design connects all 15 to the shift registers.  Are you just doing a single shift register to do 256 bytes of EEPROM space?

MarkC0x, 2020-05-20:
I wish I could say it were that simple, but I have two shift registers and it was just laziness on my part.  I was in a mental rush to show something working, so didn't connect all the EEPROM pins yet.  Then when things weren't working I went straight to code (and timing) debugging. TIL

TomNisbet, 2020-05-22:
@MarkC0x - Is everything working now?  I'll close the issue if so.
The mailman just brought me some 595s, so I'll build a version with those so I can debug everyone using the Ben Eater design,

Speaking of which, I'm building a breadboard computer right now and am dusting off this EEPROM project to do the microcode.  Expect to see a new project soon for that.

MarkC0x, 2020-05-22:
Hey @TomNisbet, yeah all is good now, so this issue can be closed (and apologies for hijacking the original issue from @8BitMark).  The 74HC595s work great as an alternative shift register.

Totally looking forward to your new computer project --that's next on my to-do list, so it would be awesome to follow along with your progress!

closed 2020-05-26 by TomNisbet