TomNisbet/TommyPROM #7: Menu appears in Arduino serial term but not in TeraTerm

8BitMark, 2020-05-21:
Having rebuilt this using 164s instead of 595s it works in serial term zap test etc.  Obviously I can't send a file that way though.

TeraTerm connects and shows the > but pressing enter does not cause menu to display it just sits there waiting.  No other command works either.  I've checked settings in Tera Term and can't see any issues correct baud, stop bits & parity etc.  

Any ideas what I might be missing?

8BitMark, 2020-05-21:
Ignore me I'm being an idiot.  
TeraTerm just needed the new line settings for send & receive changed from CR to LF under the Setup | Terminal menu.

TommyPROM now running successfully on my breadboard, written in seconds.  Thank you :-)

closed 2020-05-21 by 8BitMark

TomNisbet, 2020-05-22:
@8BitMark - someone else ran into the same issue and I never got around to putting in the one-line fix, so I'm the idiot here.  I just submitted and closed #8 to address this so now terminals will work with either CR or LF.