TomNisbet/TommyPROM #9: Schematic error

squelch41, 2020-06-13:
Thank you so much for designing this and putting it on GitHub - just built one to program a 28C512 to update a 486 motherboard bios.

There is a minor error in the schematic diagrams - the logic shifters dont have their ground and vccs connected - for me at least, not connecting these resulted in no read/writes.

Thanks again!

TomNisbet, 2020-06-13:
Hmmm, the templates I used for the shift registers don't even show the power connections.  I think that's pretty common for simple chips like NAND gates and the like.  They are probably automatically connected if the schematic is used to produce a board layout.

I'll take a look to see if that can be added to the schematic, or at least will add a note to the text about it.

Can you tell me which manufacturer name is on your chip?  There is some variation in the 28C chips by different manufacturers and it is good to know if someone has had success with a chip I haven't tried.

Glad it worked for you!

squelch41, 2020-06-14:
It was a X28C512D-15 made by XICOR
32pin DIP package

Was also able to read an old, presumably 27C512 amibios 1994 EPROM in a 28pin package (no markings on the chip itself - maybe under the label but dont want to remove)

Wired up pin 13 of the second shift register to the A15 pin

Then depending on which chip I was using, moved the VCC and connected the WE pin for the 28C chip (didnt bother for the 27C for obvious reasons!)

closed 2020-06-25 by squelch41