TomNisbet/TommyPROM #10: Errors with AT28C256

squelch41, 2020-06-24:
Hi,
I'm having problems with reliability writing to an atmel AT28C256 eeprom

About 5% of the time I can write, 95% of the time, it will fail at various points, usually below 50% but not at a consistent packet number.

Reading is fine and writing to the a xicom 28c512 is fine (i've moved the address lines to the right point for the 256).

I notice in your pcitures of your board you have some capacitors - are these to improve reliability? If so, which pins are they on?

Thanks

TomNisbet, 2020-06-24:
My current board with an Arduino Nano has .1uF capacitors from the Vcc lines of each of the chips to ground for noise reduction.  There is also a 47uF capacitor across the power and ground bus near the 5V line from the Arduino. Keep the leads as short as possible for all of the capacitors.  I'm more of a software guy than hardware, so these were just stolen from other designs rather than carefully calculated values.

I'm using my board with Atmel 28C256 eeproms right now for a Ben Eater breadboard computer and it has been very reliable.  Try to keep the wires short - my original build used pre-made jumpers and was not very reliable.

You can also try writing in byte mode instead of block mode.  It will be a lot slower, but might give different results you are just trying to get a chip burned to get on with your project.  Change the line near the top of TommyProm.ino from

```PromDevice28C  prom(32 * 1024L, 64, 10, true);```

to

```PromDevice28C  prom(32 * 1024L, 0, 10, true);```

I'll get the capacitors added to the schematic the next time I'm in there.

I hope this helps!

Tom

squelch41, 2020-06-24:
Thanks - think my problem is probably long wires and the xicom chip was just a bit more forgiving!

squelch41, 2020-07-28:
Just got a xicor X28C256P-25 - writing seems to be much more reliable that the AT28C256 - guess the xicor chips are more forgiving.

Just thought I'd post in case future people have problems!

closed 2020-08-19 by squelch41