TomNisbet/TommyPROM #12: Suggested enhancement

aemiller51, 2020-07-16:
Sometimes I only want to write a few bytes to an eeprom usually to fix a  subroutine bug, so I created a new command, POKE, syntax P SSSSS bb bb bb.. (up to BLOCK_SIZE bytes) to directly update.  

It mostly duplicates the code of the existing fillBlock command.  I had to increase the size of the CLI buffer and create a new command token as well.   

`void pokeBytes(char *& byteString)
{
 uint32_t val;
 uint32_t start;
 uint32_t byteCtr;

 enum { BLOCK_SIZE = 32 };
 byte data[BLOCK_SIZE];
 
 //first value returned is the starting address
 start =  getHex32(byteString, 0);
 
 val = getHex32(byteString, 0);
 byteCtr = 0;
 
 while ((val != 0) && (byteCtr < 32))
 {
  data[byteCtr++] = val;
  val =  getHex32(byteString, 0);
 } 

 if (!prom.writeData(data, byteCtr, start))
 {
  cmdStatus.error("Write failed");
  return;
 }

 delay(100);
 for (int ix = 0; ix < byteCtr ; ix++)
 {
  byte val = prom.readData(start + ix);
  if (val != data[ix])
  {
   cmdStatus.error("Verify failed");
   cmdStatus.setValueHex(0, "addr", start + ix);
   cmdStatus.setValueHex(1, "read", val);
   cmdStatus.setValueHex(2, "expected", data[ix]);
   return;
  }
 }
 cmdStatus.info("Write test successful");
}
`

TomNisbet, 2020-07-17:
@aemiller51 - very good!  I'm trying to get another project (breadboard computer) posted right now, but I'll revisit this when that is wrapped up - probably next week.  

For that project, I developed an eXamine command that allowed interactive listing and modification of single addresses.  The interface is like the old 8-bit ROM monitors.  You enter an address and are prompted with the current value at the address.  If you type 2 hex characters, they are stored and it moves to the next address.  A return just skips to the next address, and any other character ends the command.  It looks like this:

**X0000**
0000 FF: **01**
0001 FF: **4f**
0002 FF: **<CR>**
0003 FF: **06**
0004 FF: **.**

The result would be that the bytes at 0000, 0001, and 0003 are modified.

The downside to this is that it doesn't work well with the Arduino Serial Monitor window because that requires a <CR> before any characters are sent to the target.  On the other hand, it is really convenient when used with a real terminal.  I like your command because it can be used in any terminal environment.

Maybe I'll port them both in and we can take a look.

Thanks for the addition.

TomNisbet, 2020-08-15:
I put this on the back-burner and forgot all about it.  Checking in a change today that implements the Poke command.  Utilized the default value option of getHex32 to fix the problem of the last argument being dropped.  Thanks to @aemiller51 for a very useful enhancement to the code!

commit 885869e7 (2020-08-15): Fix #12 - add aemiller51's Poke command
closed 2020-08-15 by TomNisbet