TomNisbet/TommyPROM #15: TommyPROM v2.2 '595 Shift Register code has introduced a bug

NigelTwo, 2020-08-26:
I downloaded the V2.2 code to verify the Xmodem fix.
Some new code has been added to support '595 shift registers. I enabled the define, and verified that my hardware was setup as per the revised documentation. Yes I had used D13 as the RCLK pin too. So no wiring changes were needed.
Eeek, it didn't work.

I think the problem is this define in PromAddressDriver.cpp that is used in setAddressRegister():
`// Define masks for the address clk and data lines on PC3..PC5 for direct port control.

#define ADDR_CLK_HI_MASK    0x80`

This was 0x08 in V2.0 of the code, where no define was used. Or am I going mad?

TomNisbet, 2020-08-26:
Good catch.  Of course my quick test worked because I just zapped a pattern in that didn't cross a 256 byte boundary!

closed 2020-08-26 by TomNisbet

NigelTwo, 2020-08-26:
@TomNisbet This is a good enhancement, so I am pleased to have helped.

NigelTwo, 2020-08-26:
@TomNisbet - The conditional code for the '595 shift registers is not being included in PromAddressDriver.cpp in my environment. Have to include the Configure.h???

commit 910d07fb (2020-08-26): Fix #15 - typo in high address register pin definition.