TomNisbet/TommyPROM #16: (Minor) - Confirmation that device successfully passed the multi-pass check.

NonDualChris, 2020-09-08:
It would be nice to have an on-screen confirmation to say that is has run a successful multi-pass check and that it is OK.
(Currently, it just drops the cursor on to a new line).

Thank you so much for the project - I've successfully built the device and it works . . . ish.
Stuggling to read 27128 chips . . . bizarre problem for another time . . .

NigelTwo, 2020-09-16:
@NonDualChris, This is in response to your "Struggling to read 27128 chips" comment.
Can you describe what you have built, and what you have tried that is successful. I have built this circuit (wire-wrapped) with the intention of archiving some 2716/32/64 EPROMs. I am waiting on a "softly softly, go gently" ZIF socket as my EPROMs are physically delicate. In the meantime I think I have programmed an AT28C256 EEPROM and read it back correctly.

The TommyPROM schematic shows the 28 pin programming socket wired for a 28C256 device. But there are some pinout conflicts when 27xx(x) devices are used. To be fair, Tom has warned that the 27' device support is "Rev 0".

For the 27128 device, it looks like there might be a problem with pin1 viz.
```
     27128 28C256               28C256 27128
pin1=Vpp   A14            pin28=Vcc    Vcc
pin2=A12   A12            pin27=/WE    /P
...                       ...
```
So maybe the 27128 device will only produce data when pin1 is high. The datasheets indicate that current is drawn from this pin during a read operation.

You could experiment by either:
1. hard wiring pin1 to Vcc (easily done if you breadboarded the circuit), or
2. reading an address range that has A14 high. So instead of reading 16KB as 0000-3FFF, read 4000-7FFF. I cannot see any restrictions in the code that prevent this alternate address range.

TomNisbet, 2020-09-19:
I'm not sure what the original issue is here.  Are you referring the the Test command?

As far as 27xx devices go, you should be able to read them just fine.  The "Rev 0" support refers to some limited success I've had with writing them.

Yes, you definitely need to check the data sheets to swap around a few pins, usually the high order addresses.  The code will support address lines A0..A15 using the shift registers, plus A16..A18 wired directly to Arduino pins.  This lets you read chips as large as 512K bytes in a single operation.

You'll need to connect your Vpp pins to something.  If I remember correctly, most chips want it at Vcc in read mode, but check the data sheets.  I've used this to dump a bunch of old chips I had, including 27C040, 29C010, 29C040, 27C001, and 2764.

closed 2020-11-27 by TomNisbet