TomNisbet/TommyPROM #21: 595 shift registers and code

halak77, 2020-11-17:
Instructions say to uncomment  #define SHIFT_REGISTER_IS_595 in Configure.h 
I do not see this in Comfigure.h nor do I quite see how it requires this when I look at. The adrressing code ( it appears that the 595 rclock is toggled whenever an address has been shifted.  However my attempt of building programmer does not seem to be updating the address (writes and reads one value to entire room.
Thanks for any help.

TomNisbet, 2020-11-17:
I removed the conditional code as part of the fix for #18 but missed that comment that mentions it. I’ll update the code to remove comments a that reference it. 

The code should work with either type of shift register without modification. If your addresses aren’t updating you can try the hardware verification sketch to help track it down. Also, be sure to ground the OE pin on the shift registers. On some datasheets this pin is marked as G instead of OE. 

Are you sure you are writing the same value to the entire ROM? Or does it just look that way because the addresses don’t change?  What it the output if you do a Z0 command followed by D0?

halak77, 2020-11-17:
That is how the code appeared to me ( no conditional) I am starting to think my problems are the use of an old prototype board, that and my workbench is to cluttered to work there ( with o scope) I found that I had the rclk line connected to the srdata line. After I fixed that ( and checked my work once again, I still it writing same data everywhere. I am not at home, but I am guessing I missed the OE on the s/r—- doh! Thanks for help. I will let you know how it works out. Thanks also for your efforts.

Don Peddicord

halak77, 2020-11-27:
Tom, 
Again thank you for shariig your code. I went to a soldered build on a proto board. Once I solved the bad solder joints, I still had trouble . It turned out that I needed pull ups on the shift register reset lines (I would have paid to read the data sheets, I thought these were same as to but I was wrong, unconnected inputs do not pull themselves up!
Again thanks for your code and your help.
Best regards,
Don Peddicord

TomNisbet, 2020-11-27:
Don,
I'm glad its working and that you have found this project helpful.  I've been helped by countless projects that other have put up and it's nice to know I've managed to give one back.  

I updated the documentation to mention the RESET pin on the shift registers.  If the update still isn't clear or you see anything else that might save time or frustration for the next person, please let me know.
Tom

closed 2020-12-21 by TomNisbet