TomNisbet/TommyPROM #23: can it run in arduino mega? because mega has alot of gpio, no need other ic.

youxiaojie, 2021-03-08:
can it run in arduino mega? because mega has alot of gpio, no need other ic.

TomNisbet, 2021-03-08:
There is no current version for the mega, but the code is very modular and it wouldn't be too difficult to make a mega version.

There is some low-level code written specifically for the Uno/Nano to meet the timing requirements for the EEPROM chips.  To run on a mega, you'd need to modify the port-specific code in PromDevice.cpp to match the mega ports.  You would also need a new version of PromAddressDriver.cpp, which would be much simpler than the original one because it would just need to put the address out on some pins instead of driving the shift registers.

youxiaojie, 2021-03-08:
1 the specific code is mainly in adress driver? which needs time sequence? 
2 is there difference between 27/28/29/39 series of eeprom/flash?

TomNisbet, 2021-03-09:
The address driver contains the code that sends the EEPROM address using the shift registers.  You could replace this with code that just writes the address directly to the appropriate pins on the Mega.

There shouldn't be any timing dependencies when reading or writing the chips if you are doing single byte writes.  The chip-dependent timings happen when you are using block write modes.  Specific timing is also needed when sending SPD unlock codes, like with the 28C families.  If you set the block size to zero when declaring the target chip, then block writes won't be used and there shouldn't be any timing-specific code.

With that said, you will probably need to make some adjustments to the code in PromDevice.cpp.  There are routines there, like readDataBus and writeDataBus, that use direct port access to PORTB and PORTD instead of using a set of the much slower digitalWrite calls.  Unless you can use the same ports on the Mega that were used on the Uno, those calls won't work without some modifications.

Good luck with your project.  I have a Mega here, but never tried to use it for TommyPROM.  There is no reason it wouldn't work, and the design would definitely be much simpler.

wpcarro, 2024-06-09:
@TomNisbet would you be open to a pull request if I give this a try?

TomNisbet, 2024-06-09:
It may be difficult to come up with a clean way to have both the mega and the nano/uno in the same code base. It feels like there would be a lot of #ifdefs, but maybe it will be cleaner than I’m imagining.

TomNisbet, 2024-06-09:
On second look, maybe not too bad. It looks like the direct hardware writes are just in the PromDevice base class and in the address driver. If you are able to keep the hardware using the same Arduino pin names for the control lines, i.e. WE, CE, and OE, then the changes might be fairly localized.

I don’t have access to the code for a few days, so I’m only able to peek at the repo on a tablet. I may be missing something.