TomNisbet/TommyPROM #36: 5.5 volts , USB volts , VCC volts

haroldmoore, 2021-12-21:
There is 5.5v marked on the Nano, what is VCC then? I'm thinking on getting two little charge pump boards for the 5.5 an 25.5 volts
no?

here is a video I made -----> https://youtu.be/NpmLo4B8duw

Thanks in advance!

TomNisbet, 2021-12-21:
Sorry for the confusion.  I'm not sure why the library labels that Arduino pin as 5.5V.  It is labelled as 5V on the silkscreen of my Unos and Nanos.  Vcc should be 5V and it should connect to both the Arduino's 5V pin and the 8755's Vcc.

The data sheet says that the 8755A wants Vcc between 4.75V and 5.25V.  The Vdd pin should be at 25V for programming and 5V (Vcc) for reading.  Because the circuit that switches the voltage on the Vdd pin will drop a fraction of a volt through the diode, I fed it 5.5V and 25.5V from a bench power supply.  I very rarely need to program an 8755, so this was convenient for occasional use.

You could *probably* get away with using the same 5V input as both the Vcc and the low voltage input to Vdd.  The voltage drop depends on the diode you use.  

Similarly the programming voltage needs to be 25V +/- one volt, so if you have a convenient 25V source, then the switcher is going to drop it less than one volt and you'll still be in spec.  Because I had an adjustable supply, I just dialed in the two Vdd voltages until the switcher output gave me the ideal 5V and 25V.

Good luck with your project and feel free to reach out if you have any questions.  Are you building and 8085 system?  That was the brains of my first-ever computer.

-Tom

haroldmoore, 2021-12-22:
Tom,
          Thanks for clearing that up for me! I've ordered a couple of
adjustable charge pump modules to
           finish the project for my intel SDK-85 board, I'll have two
programmers, one for the second position
           8755A and the other for a large flash prom that uses the
extended memory drivers. The issue is getting
           programs into the system fast enough and still be able print out
on my 110 baud ASR33 teletype.
                     An educational system for the kids at school maybe,
because there are a LOT of  beginning curriculums
           on the 8085, say could you point me to where I could change the
memory printouts to be all upper-case in
           the code?

           Thanks in Advance!

           Harold

TomNisbet, 2021-12-22:
Do you want to uppercase the hex dump or something else?  If it is the hex dump, just change the hex[] array in TommyPROM.ino around line 72.

haroldmoore, 2021-12-22:
Tom,
           Ah great that fixed the hex dump! ok just waiting for the charge
pump modules to
           test the programming, I put an 8355 in and it read this ok as
well, don't have any to
           program since there OTP. ok I put a note at the top comment
section, here are some pictures:

closed 2022-12-17 by TomNisbet