TomNisbet/TommyPROM #37: Programming a 27C040 - will this work

danriches, 2022-01-07:
Hi Tom,

I would like to program a couple of 27C040 EPROM's and would like to know if your programmer will work? I see in the code for the 27 series that there's a write function hence why I asked as it's not stated in the readme that it does yet.

Kind Regards, Dan

TomNisbet, 2022-01-20:
Dan,

The 27C write algorithm is fairly straightforward, but I don't remember if I've actually programmed any chips with it.  The bigger issue is that these chips require a 12.75V on Vpp, although it doesn't need to be switched on and off.  The Fairchild data sheet I have also mentions using a 6.5V for Vcc while programming.  I'm not sure how strict that is and if you could get away with using 5V.  I'll have to see if I can dig up a better datasheet for the chip, because the one I have has the Vpp and Vcc voltages swapped in some parts of the documentation, so I'm not completely confident it is correct.

With all of that said, I suspect that the code would work if you can wire up a power supply for the Vpp and are careful to apply the voltages to Vcc, Vpp, and PGM in the right order.  

I have an NM27C040Q in the parts bin here, so I may wire it up and try it.

5arid, 2022-09-01:
to program 27C0xx EPROMs you can use this :
https://forums.nesdev.org/viewtopic.php?p=256197#p256197

wjhun, 2023-07-15:
Just a note that I managed to successfully write to an AM27C040 device using a TommyPROM, with only a slight change to the PromDevice27 line used for this device (see #57).

Thank you, @TomNisbet, for this fine project!

[image]

TomNisbet, 2023-07-21:
Glad it was helpful!  Thanks for the update.

closed 2024-05-29 by TomNisbet