TomNisbet/TommyPROM #45: W27C512

Henkvd72, 2022-10-02:
Hellom love the project, i am builing one but a have a question, 
can it programm the Winbond W27C512?

TomNisbet, 2022-10-02:
Good timing on your question.  I have not tried that chip, but I have some Winbond 27C257s on order which use the same algorithms.  These do need 14V to erase and 12V to program, but they don't require a a pulse at this voltage, so it should work to just apply the voltage manually as described in the [27C-EPROM](https://tomnisbet.github.io/TommyPROM/prom-families/#27c-eprom) section.  They chips should be in next week, so I'll post an update after I've had a chance to look at them.  The current 27C code does not implement the erase command, so that will need to be added.  There may also be some changes to the programming algorithm if it is significantly different from the EPROM versions.

TomNisbet, 2022-12-17:
The 27C257 code has been tested and checked in, so I suspect the W27C512s will now work with that.

closed 2022-12-17 by TomNisbet