TomNisbet/TommyPROM #53: 27C256

ElectroBoy404NotFound, 2023-03-14:
Hello,
How to use the code for 27C256?

TomNisbet, 2023-03-14:
The 27C256 chips vary a bit, depending on the manufacturer, but the existing code should work for you.  Make these two changes to the default software:

- In Configure.h, comment out the #define for PROM_IS_28C and uncomment PROM_IS_27
- Near the top of TommyPROM.ino, uncomment the PromDevice27 declaration for the 27C256 and make sure all of the other declarations in that section are commented out. 

On the hardware side, you will need to apply a programming voltage from an external supply to Vpp to write the chip.  If you are just reading the chip, then you can tie Vpp to the +5V.  Check the data sheet for your chip to verify the programming voltage.  I have seen some that want 12.75V and others that want 13V.  The 12.75 chips are +/- .25 volts, so 13V is probably good in either case, but don't exceed that.

Please let me know the full number on your chip, i.e TMS27C256-15, so I can add it to the list of supported chips.

ElectroBoy404NotFound, 2023-03-30:
> The 27C256 chips vary a bit, depending on the manufacturer, but the existing code should work for you. Make these two changes to the default software:
>
> * In Configure.h, comment out the #define for [...]

Hello,
My chip number is "M27C256B-12F15880X1140L"

TomNisbet, 2023-03-30:
Were you able to program it successfully?
Your chip is an ST Micro EPROM.  It needs 6.25 volts on Vcc and 12.75 volts on Vpp when programming.  See page 8 of the datasheet here: https://media.digikey.com/pdf/Data%20Sheets/ST%20Microelectronics%20PDFS/M27C256B.pdf

ElectroBoy404NotFound, 2023-03-31:
> Were you able to program it successfully? Your chip is an ST Micro EPROM. It needs 6.25 volts on Vcc and 12.75 volts on Vpp when programming. See page 8 of the datasheet here: https://media.digikey.com/pdf/Data%20Sheets/ST%20Microelectronics%20PDFS/M27C256B.pdf

I'm going to test it today.

ElectroBoy404NotFound, 2023-03-31:
@TomNisbet 
**UPDATE:** I tried 2 EPROM models. I was able to prpgram the M27C256B-12F1**B880C** chip fine but not the M27C256B-12F1**5880X** chip. I think the problem is that my 12v adapter is actually giving around **11.5V** instead of **12.75V** needed for programming and my tab (Samsung A8 that I'm using to program it) is giving around **4.95V** instead of **6.25V** needed. Is there any way to solve these problems (without replacing the power supply)?

TomNisbet, 2023-04-01:
The minimum voltages for that part are 6V and 12.5V, so you might get lucky with some chips, but your results won't be reliable without a different power supply.  The good new is that you have verified that your programmer is otherwise working.

Also, it you get an external supply for the 6.25V to your EPROM's Vcc, make sure that it is not connected to the 5V from the Arduino or the 5V going to the shift registers.

ElectroBoy404NotFound, 2023-04-03:
> The minimum voltages for that part are 6V and 12.5V, so you might get lucky with some chips, but your results won't be reliable without a different power supply. The good new is that you have [...]

I see. I will try using a boost converter module for 6.25V and see if it works. Also, sorry for the late reply, I was busy.

ElectroBoy404NotFound, 2023-05-07:
I tried it, it did;nt work

ElectroBoy404NotFound, 2023-05-07:
I tried 13v and 7v, but it just would'nt program.

jscrane, 2024-05-03:
I have a couple of similar chips which I can't program either. They are labelled ```M27256F1```.

I am using a TommyPROM32 board with Vpp at 12.5v and Vcc at 6.5v.

Referring to [this datasheet](https://www.silicon-ark.co.uk/datasheets/m27256-datasheet-st.pdf), shouldn't the ```pulseWidth``` constructor parameter be ```1000L``` maybe? (In any case, I tried both 100L and 1000L and neither worked.)

> The duration of the initial E pulse(s) is 1ms, which will then be followed by a longer overprogram pulse of length 3ms by n
> ...

Happy to help debugging this problem...

TomNisbet, 2024-05-04:
I have programmed a M27C256, but yours are the older M27256 with a different programming algorithm.  It looks like your data sheet wants 6V on Vcc  instead of 6.5.  Hopefully that didn't cause any issues.

The program pulse width should be 1000L and the overwrite value should be 3.  I checked in new code that added a definition for this chip in TommyPROM.ino.

The existing code was also incorrect.  The overwrite pulse should be programPulse * multiplier * numberOfWriteAttempts.  I was not including the multiplier for the number of attempts.  Also the code was not doing the overwrite pulse for chips that do not have a dedicated WE pin. 

Please let me know if the new code works for you.  I'll add the M27256 to the list of verified chips if it works.

jscrane, 2024-05-04:
I can confirm that the latest code works with my M27256. Thanks a lot!