TomNisbet/TommyPROM #54: Hardware Verify Not Working?

OrlandoBlyat, 2023-04-08:
Hi,
I've assembled the TommyPROM on a breadboard with the 74HCT595 registers. But writing/reading aren't functioning properly, my wiring probably is at fault. I uploaded the HardwareVerify sketch, but in my serial terminal (MobaXTerm) I don't have a CLI (I set the baud rate to 115200). I recieve two '#' and that is it. The standard TommyPROM sketch works as inteded. Screenshots present. 
[image]
/Terminal with the Hardware Verify Sketch/

[image]
/Terminal with the TommyPROM sketch/

TomNisbet, 2023-04-10:
The verify sketch doesn't print any header to identify itself, just the # character for the prompt.  Did you try entering any commends?  It uses the same terminal ode as TommyPROM, so if you were able to use one, then the other should work as well.

What kind of chip are you programming?  Can you try the Z command in TommyPROM and post the results?  Do Z0 and then D0.

OrlandoBlyat, 2023-04-17:
Hello,
sorry for the late response, I guess GitHub didn't send me an email about the response.
As I said, the terminal is exhibiting kind of strange behavior:
[image]
----
[image]
----
I can input the command, pressing enter returns the cursor to line start. None of the commands return anything.
[image]
----
I've also tried using TeraTermYT but it doesn't change.

[image]

SvenPaterson, 2023-10-20:
I too seem to have had this exact same issue using teraterm. Worked just fine in the arduino native serial monitor once I finally tried using it!

closed 2024-05-29 by TomNisbet