TomNisbet/TommyPROM #61: Cannot burn image

ian-rose, 2024-02-29:
I am trying to burn an image to a 28C256, but it is not working.

MacOS Sonoma 14.3
minicom version 2.9

The commands I am running:

1. `minicom`
2. `w`
3. `esc-z`
4. `s`
5. `down`, `down` (select xmodem)
6. `enter`
7. `left`, `left`, `left`, `left`, `left` (select `Goto`)
8. `enter`
9. (type in directory)
10. `enter`
11. `down`, `down`... (highlight file)
12. `space` (select file)
13. `enter`
14. wait (window shows, but no progress indicator)
15. `d`
16. `enter`

The dumped data is not correct

Based on my ability to disable write protection I assume things are wired correctly. I can think of two possible issues:

1. My image is not correct
   - My image is a binary file with no header information. When viewed with `xxd` the contents are correctly displayed.
2. I am not executing the correct commands
   - I have yet to find any documentation as to how to actually burn an image using TommyPROM, so this is very likely.

TomNisbet, 2024-03-01:
There are a few things to try:

- Have you verified that the basic programming functionality works?  Use the Zap command, z0, to verify that a test pattern can be burned into the chip.  If that fails on the first byte, you may need to use the Unlock command as well.
- A binary image file is the correct format to use, so no problem there.
- Use the w0 command to specify that the image should start at the first byte of the chip.
- If the Modem transfer fails, the error message is sometimes hidden by the transfer.  After you get back to the command prompt, use the '/' command to show the last error message.  That may give you an indication about the failure.
- Also try the Info command, and post the results of that here.  It will show detailed information about the last transfer.

When you say that the dumped data is not correct, are you seeing that the data in the chip changed after the write but it isn't the data you expect, or that no new data was written at all?

ian-rose, 2024-03-11:
I was able to verify writing via successfully unlocking the chip. I did not try using zap.

I found a workaround to this issue by installing lrzsz and configuring minicom to run the following command when xmodem is selected: `lsx -b -X`. I'm assuming this is very kludge-y (xmodem should work without overwriting the command it executes), but it got the job done.

closed 2024-03-11 by ian-rose

ABelliqueux, 2024-06-14:
To add to what @ian-rose said, I also had to install `lrzsz` on Manjaro Linux but I had to use `lrzsz-sx` instead of `lsx` in minicom.

Here are the steps to change the default xmodem command to `lrzsz-sx -b -X` :

In minicom:

1. Ctrl-A, Ctrl-Z
2. Hit O key
3. Enter "File Transfer protocols"
4. Edit line with "xmodem", making sure it's the send command ( should say "sx" and not "rx") or hit C key.
5. Hit Return once and edit the existing command (`lrzsz-sx -vv` in my case) to `lrzsz-sx -b -X`
6. Hit Escape a few times until you're back in the configuration menu.
7. Choose "Save setup as dfl"

Maybe this could be mentionned in the doc ? I can submit a PR if you want me to @TomNisbet.