TomNisbet/TommyPROM #66: CSI CAT28C16AP working?

jscrane, 2024-04-02:
Hi,
Was wondering if anyone has tried this chip? I have added a line to `TommyPROM.ino`

```
PromDevice28C  prom(2 * 1024L, 0, 10, true);  // 28C16 with no page writes
```

However I'm not having any luck writing to it:

```
>u
Writing the unlock code to disable Software Write Protect mode: OK
>p 0 00
INFO: Poke successful
>d 0 0f
00000: 00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  ........ ........ 
>p 0 ff
ERROR: Write failed
>d 0 0f
00000: 1c 1c 1c 1c  1c 1c 1c 1c  1c 1c 1c 1c  1c 1c 1c 1c  ........ ........ 
```

I'm talking to it using minicom on Linux.

(I have assembled TommyPROM on a PCB btw.)

Any help appreciated!
Steve

TomNisbet, 2024-04-02:
Hi Steve,

I have not used the CAT28C16, but have programmed the CAT28C256.  There is no reason the code shouldn't work with that chip.

The 28C16 does not support Software Data Protection, so the unlock command isn't doing anything.

It's odd that all of the bytes change between your two read commands when you only write one byte.  That makes me suspect that the addressing circuit may have a problem and is possibly just reading a single byte over and over.  It could also be that it isn't reading data back from the chip at all and those values are just random signals from the bus.

You could try the Hardware Verify Sketch described on the [Troubleshooting page](https://tomnisbet.github.io/TommyPROM/docs/troubleshooting) to see if the hardware is behaving as it should. 

If you have a second chip, especially a different type, that would be helpful to test as well, to isolate if there is a hardware problem or if you just have a faulty chip.

Your setup for the chip looks good.  Byte at a time with no block writes is the way to go.

Did you design your own PCB or did you use one of the designs on my site?

jscrane, 2024-04-02:
Hi Tom,
I used your PCB designs. This one is the plain TommyPROM one, I haven't finished the TommyPROM32 yet.

This chip only has 24 pins --- does that board only handle 28 pin devices? 

I have done a quick check of the address lines with the Hardware Verify sketch and a multimeter and they seem OK --- for a 28 pin device.

I guess I will have to wait for the rest of the parts to arrive and complete the TommyPROM32 board?

Thanks!
Steve

jscrane, 2024-04-02:
Looking at the pinouts for the 28C16 and the 28C256, I guess I could put the chip with its pin 1 in pin 3 of the ZIF, and connect:
- ZIF pin 28 to pin 26 (Vcc)
- ZIF pin 27 to pin 23 (/WE)

All of the other pins seem to be the same. I will try this tomorrow and report back here.

TomNisbet, 2024-04-02:
The problem you will have there is that you will be shorting the A13 pin from the shift register to Vcc and the A11 pin to WE.  If the A13 pin is low then you have a short circuit.  Similarly, the WE and A11 signals will conflict with each other.  If I ever make a new version of that board, I should add jumpers for those two pins to accommodate both the 24 and 28 pin chips.

The TommyPROM32 board would work very well.  There is already a jumper position to put Vcc on pin 24, so you'd just need to add a single jumper wire for WE and then just install all of the other jumpers in their default position to connect all pins straight across.

jscrane, 2024-04-02:
Yes you are right of course. I will wait for TommyPROM32!

Thanks for your help!

closed 2024-04-02 by jscrane