TomNisbet/TommyPROM #72: TommyPROM32 : Ok to replace SW1 and SW2 with jumpers ?

ABelliqueux, 2024-06-15:
As switches with the correct size/footprint  are hard to source at a decent price where I am, and only one of the position is used, is it okay to replace them with a jumper like so ? 

[image]

[image]

[image]

TomNisbet, 2024-06-19:
Yes, that should work just fine.

ABelliqueux, 2024-06-19:
Cool, thanks !

closed 2024-06-19 by ABelliqueux