TomNisbet/TommyPROM #78: Some files won't burn

micha-uhl, 2025-01-02:
Hello, I have the problem that some files can not be burned or crash after a few blocks. other similar ones work. why is that?

TomNisbet, 2025-01-02:
There shouldn't be any data-specific differences, but maybe there could be a loose or incorrect data or address connection to the EEPROM that only shows up with specific data.

Can you post the output of the Write command for a failed file followed by the output of an Info command?

micha-uhl, 2025-01-02:
Hello, I think it's because of the EEPROMS. They are seeq DQ28C256-250. Also with the apparently working file come errors, whole blocks are falsified. See picture, should everything be FF.
[image]
I ordered others.
regards Michael

TomNisbet, 2025-01-02:
I've never tried that particular chip, but the data sheet looks like it would be compatible.  Two things you might want to try:

In TommyPROM.ino, find the PromDevice64 line that isn't commented out and replace the 64 with 0.  This will disable page writes.  If that works then there may be a timing issue or something else that is broken in the block write code.

You can also try the Zap command to see if it writes correctly on the page boundary.  It would be interesting to see that same memory block after these three commands:

- Z 5000
- Z 5030
- Z 5060

micha-uhl, 2025-01-03:
Hallo Tom, I tried your Suggestions, but In both cases it now aborts at 3F00, with both EEPROMS!
The Z command works.
[image]

[image]

[image]

micha-uhl, 2025-01-11:
Hi, now I bought new AT28C256F 15PU. With or without page write, the first bytes up to $80 are misspelled. With the "Z" test it works. The rest of the file appears to be error-free. 
[image]
Do you have another sketch for only 28C EEPROMs?

ABelliqueux, 2025-02-21:
@micha-uhl : Are you still experiencing issues ? Can you tell us which OS and serial terminal software + version you are using ?